enactic/openarm #53: OpenArm End Effector Improvements

edwin-giang, 2025-02-25:
**Action Items for End Effector Improvements:**
- [ ] Strengthen wrist to hand attachment method
- [ ] Reconduct risk assessment for the wrist
- [ ] Change gripper to become less ARX-like
- [ ] Add backdrivability to claw (less friction)
- [ ] Redesign leader controller

hiroyams, 2025-02-27:
low priority, minimal fix:
- safety and wrist

Ravoik, 2025-03-06:
[image]

[image]

Finalized for v0.2, printing to test

Ravoik, 2025-03-06:
[image]

edwin-giang, 2025-03-18:
03/17
---
Breaking leader controller into separate sub-issue. For backdrivability, increased tolerance between bridge and rail -- using silicone spray to reduce friction of parts. End effector redesign is complete for v0.2, closing issue.

closed 2025-03-18 by edwin-giang