enactic/openarm #55: OpenArm Simulation + URDF Improvements

edwin-giang, 2025-02-25:
**Action Items for Sim + URDF Improvements:**
- [ ] Update how left + right jaw joints are defined in simulation
- [ ] Fix base position
- [ ] Generate URDF for Pedestal
- [ ] Define force actuated limits
- [ ] Define inertia and mass of joints
- [ ] Create a better URDF w/ simplified geometry and no fasteners
- [ ] Define self-collision
- [ ] Create URDF w/ 5DOF EE attachment
- [ ] Create simple script to swap in different EEs
- [ ] Create more examples for simulation (IL, RL)

**NOTE:** Will most likely need to break this down further into sub-issues

1lokeshpatel, 2025-02-25:
URDF R/L Joint DOF Comparison:

- **joint 1**: same
- **joint 2**: mirror
- **joint 3**: mirror
- **joint 4**: same
- **joint 5**: mirror
- **joint 6**: same
- **joint 7**: mirror

edwin-giang, 2025-03-05:
03/05 Updates
---
- Generated URDF + mesh for pedestal
- Simplified URDF w/o fasteners, no rack on jaw joints
- Used RobotCAD to make conservative blocks for collision
- Used mimic to link left and right jaw

edwin-giang, 2025-03-05:
Need to merge pedestal v0.2 mesh and openarm mesh

Ravoik, 2025-03-05:
Edwin creating rigid groups in CAD with bodies to simplify geometry and export unified meshes for joints.

edwin-giang, 2025-03-06:
Waiting on left arm CAD from @Hiroki-HANAMORI before creating URDF export version

edwin-giang, 2025-03-10:
03/08
---
Created left arm CAD based off the assembly given by Hanamori-san. Closing issue since all CAD improvements for URDF v0.2 are completed

closed 2025-03-10 by edwin-giang