enactic/openarm #59: Left Arm Design

edwin-giang, 2025-02-25:
**Action Items:**
- [ ] Adapt mechanical design for left OpenArm

Ravoik, 2025-03-05:
- Ordered some flipped parts for Open Arm left arm
- @Hiroki-HANAMORI need to discuss the shoulder joint parts, need to flip and order LHS versions of these parts because the limits are different from the right.

edwin-giang, 2025-03-06:
03/06
---
Determined plan to create left and right arm using mainly pre-existing parts. Left arm joint 2 will have bearing located at front, right arm will have bearing at back. For both arms, follow the guide below for assembly:
[image]

edwin-giang, 2025-04-04:
03/19
---
Completed reassembly of OpenArm for left arm. Ordered mirrored sheet metal parts for left wrist and assembled. Closing issue under v0.2, will iterate in future versions

closed 2025-04-04 by edwin-giang