enactic/openarm #63: Strengthen wrist to hand attachment method

edwin-giang, 2025-02-28:


edwin-giang, 2025-02-28:
Assigned Lokesh to issue

1lokeshpatel, 2025-02-28:
finished first draft of design, located in Fusion360. ReazonResearch --> Triad --> Data_by_LOKESH --> EndEffector_WristRedesign

**Changes to make:**
- add counter sink for screws into motor (reducing size of that plate
- shorten link as much as possible

1lokeshpatel, 2025-03-05:
## 03/05 Update

- added countersunk holes for both plates
- added fillets to smoothen design
- reduced plate size and link length size 
- changed to cap design instead of small juts for hardstop

[image]

1lokeshpatel, 2025-03-06:
## 03/06 Update

- Printed first design

**Issues:**
- the 4th attachment hole for end effector motor cannot be blocked by linkage as it is needed to strengthen and ensure it doesn't snap
- slight tolerancing changes needed for countersinks and cap design to ensure there are no collisions with fasteners
- more feature fillets

**Fixes:**
- create the feature fillets
- widen the linkage as much as possible and create fillets where possible for stress distribution

1lokeshpatel, 2025-03-10:
## 03/10 Update

Tested with feature fillets and 4th hole addition. Works okay but need larger opening for that screw to be inserted in a more smooth manner. Also printing in a different orientation helps with overall print quality.

[image]

**Changes:**
- larger opening for that screw hole, increase overall diameter of pipe extrude cut but also expand horizontally out of the plate.

1lokeshpatel, 2025-03-11:
**Final Iteration, will reopen if testing reveals flaws:**

[image]

closed 2025-03-11 by 1lokeshpatel