enactic/openarm #64: Define inertia and mass of joints

edwin-giang, 2025-02-28:


edwin-giang, 2025-03-04:
03/04 Meeting:
---
- Currently testing one arm for inertia values
  - Had issue with baudrate, now reset so testing can continue

thomasonzhou, 2025-03-04:
Test code available at: https://github.com/reazon-research/inertia-test
The baud rate update python script is reset after power cycling, so we should opt to flash permanently when possible (@1lokeshpatel do we have another open issue already?)

edwin-giang, 2025-03-06:
Waiting to test 4340P motors, blocked since motors have not been assembled -- devising new test plan

edwin-giang, 2025-03-11:
03/11
---
Opened issue #78 for 4340P motors, closing this issue now as all inertia testing for 4340 is complete.

closed 2025-03-11 by edwin-giang

thomasonzhou, 2025-03-12:
Masses (kg)
1. 0.486
2. 0.178
3. 0.404
4. 1.123
5. 0.558
6. 0.470
7. 0.346
8. 0.298

*Link 1 did not include the the mounting components:

[image]