enactic/openarm #77: Create ROS2 control library

edwin-giang, 2025-03-06:


edwin-giang, 2025-03-06:
Moved issue to vFuture

thomasonzhou, 2025-03-06:
WIP on ros2_control branch:
https://github.com/reazon-research/openarm_ros2/tree/ros2_control

Subtasks:
- [x] define control interfaces
- [  ] complete hardware package (start with position control, then velocity, torque)
- [  ] create simple controller (PID)
- [  ] complete bringup package (1 command launch)

edwin-giang, 2025-04-04:
moving issue to v1.0

edwin-giang, 2025-04-10:
Video of ROS2 Control implementation

https://github.com/user-attachments/assets/f9a173ba-f2f1-4ecc-b61e-a9716bf19885

thomasonzhou, 2025-04-10:
Merged to main https://github.com/reazon-research/openarm_ros2/tree/main

Requesting a commit bump in the main repo @edwin-giang 
- https://github.com/reazon-research/OpenArm/pull/92

closed 2025-04-24 by hiroyams