enactic/openarm #78: 4340P CAD and URDF Adjustments

Ravoik, 2025-03-06:
We need to adjust the CAD caps that interface with the 4340P as well as re-measure the inertia values for the effected axis.
- [ ] 4340P CAD modified
- [ ] 4340P parts ordered
- [ ] 4340P mounted on arm and inertia measured

Hiroki-HANAMORI, 2025-03-14:
[image]
CAD updates have been completed to accommodate 4340P.
The general shape of the arm has changed little.

Left side of image: old arm
Right side of the image: arm with 4340P installed

Hiroki-HANAMORI, 2025-03-14:
[image]
[image]

The cross roller version of the DAMIAO gear head has been modified to a structure that holds the bearings in place.
Therefore, it was modified so that the cap for attaching the key can be assembled without disassembling.

J1 has already been assembled with Ory-san's arm, so no change is necessary.
J3 was modified to install the gear head without disassembling the gear head, reflecting the reflection on Ory-san's arm.

Hiroki-HANAMORI, 2025-03-14:
Preparation of order data has begun.
Inquiries are underway for J3 caps due to a processing error.

To avoid errors in the quotation, I am also preparing data with the error points removed, but this will reduce the strength of the part.
If the quotation takes time, we will proceed with the order.

edwin-giang, 2025-04-04:
04/03
---
Moving issue to v1.0. New parts for 4340P arrived and reassembly was completed. Will adjust and reorder parts as needed

Hiroki-HANAMORI, 2025-04-04:
Details of any problems identified in assembly and plans for revision

- Axis parts for symmetrical parts, length needs to be extended by 1,mm.
-- Perhaps the fact that the 4310 was made from the CAD of the 4340 had a negative effect.
-- Despite the different thickness of the TOP caps of 4340 and 4310, they were designed with the same dimensions, which caused the axis to sink 1mm...
-- In this case, we used a 1mm spacer
- The shaft part coming out of J1 needs to be extended by 3mm in length.
-- Purely an ordering error.
-- The old data was submitted.
-- This time, we used a 3mm spacer.
- Connector position of 4340P is not very good.
-- The connector position was subtle because the original hole on the top surface of the motor was used.
-- The base part could be handled by changing the slit position.
-- J3 may have to allow the connector to be tilted.