enactic/openarm #81: Autocalibration on Startup

thomasonzhou, 2025-03-07:
There is currently a gap between CAD/URDF/MJCF joint parameters and the real joint parameters on real robots, namely:
- joint limits
- zero positions

To add an additional layer of reliability on top of the mechanical hard stops, it would make sense to add software stops. One way to do this would involve:
1. detecting the true joint limits on startup (gently move position command to ends until it does not change)
2. moving to a known offset (desired zero position)
3. set zero positions of motors

edwin-giang, 2025-03-18:
Moving to vFuture

edwin-giang, 2025-04-04:
Assigning issue to @Ravoik, moving issue to v1.0

Ravoik, 2025-04-14:
Measured joint limits and necessary offsets, converting to rads.
Figured out MIT control and how to use the library.
Wired spare open arm, setup with pedestal. 
Finished 1st version of script, need to clean up and comment on code. Will post a video here.