enactic/openarm #90: Camera Integration w/ OpenArm

edwin-giang, 2025-04-04:
*What is the pre-existing problem? Please describe.*
Need to add a camera to OpenArm + Pedestal to collect date for RL, IL, and other training

*Describe the solution you'd like*
Mount a single realsense camera to the top of OpenArm pedestal

*Next Steps*
- [ ] Design a camera mount to attach realsense camera to pedestal

edwin-giang, 2025-04-04:
04/03
---
Updates from 03/27 to 04/03:
Designed single camera mount to attach realsense cam to openarm pedestal. Made tilt adjustable to optimize angle and FOV.
[image]

Using this setup to train OpenArm until 04/07, will update camera mount design afterwards and discuss if additional setup is needed

thomasonzhou, 2025-04-10:
Used in GR00T N1 initial data collection and deployment: https://github.com/reazon-research/Isaac-GR00T

Seems to be a flexible enough solution for now, closing.