enactic/openarm #96: April Meeting Minutes

edwin-giang, 2025-04-15:
April Meeting Minute Documentation

edwin-giang, 2025-04-15:
04/15 RD Meeting Minutes:
---
- RT Meeting:
  - Updated RT on feedback from US
  - got manufacturing feedback from RT
  - shared timeline for openarm
  - do beta release before v1.0
  - will take responsibility for manufacturing + shipping
  - can do shipping overseas
  - RT will take responsibility for custom pieces
- Edwin:
  - Covers
  - Leader Controller
- Ryan:
  - Getting motors
  - Auto-calibration
- Lokesh
  - tested control frequency for 1Mbps (6-8 motors)
    - 6 motor control frequency -- 1.175kHz
    - 7 motor control frequency -- 1kHz
    - 8 motor control frequency -- 900Hz
  - Changed to 5Mbps
  - Changed STM32 code to run at 5Mbps
    - 8 motor control frequency -- 1.6kHz
  - Added save motor param for baud rate -- permanently sets baudrate
  - Planning to add synchronization
  - Will document and onboard Toki-san on STM control
- Hanamori:
  - Shortened length of forearm
  - Adjusted limit of elbow
  - Will arrive 04/21
  - Updated risk assessment guideline
  - Need to update BOM
  - Need to record dimension of cables, where to purchase, installation
- Other Items:
  - OSS Project (Pollen)

edwin-giang, 2025-04-17:
04/17 Meeting Minutes
---
Ryan:
- DM8009P Motors Arrived
  - No 8009 motors :(
- Unified and commented on calibration script code

Toki:
- Setting motor IDs and verifying the motor firmware
- Testing bilateral feedback on 5Mbps using STM32

Edwin:
- Updated leader controller
- updating covers

Lokesh:
- Improving STM32 library functionality (by Friday)

Thomason:
- MoveIt pick and place is priority
- Mujoco can be done remotely, will be deprioritized 

Hanamori:
- Assemble 8009P motors into openarm

Iida:
- Making cables for 8009P motors
- Testing 48V AC adapter using load tester

Pruthak:
- Doing research for next phase of openarm
  - Moving base, other items
- Working on assembly manual, auxiliary items
- Will work with hanamori-san to assemble arms

hiroyams, 2025-04-24:
# OpenArm v0.3 Feedback
## Other
- move important logs in issues to main doc
  - links to all files + documentation per subproject
- Safety doc 
## Hardware

### In Progress
- 8009 accommodation
  - eval of 8009
  - redoing harnessing
- Safety TPU covers
- Linear EE
  - make some 3D printed parts to be machined
- LCSC connectors are mid
  - documentation: still in issue
  - [ ] see other connectors from LCSC
  - [ ] investigate other supplier
- Torque testor
  - MyActuator
  - RoboStride
  - WinTech
- Static discharge investigation
- Camera mounting options
- Wrist design
  - Inspire-hand mount
- Disassembly Procedure for DAMIAO
  - want to fix before moving to production w RT
### Future
- Wrist redesign
- Simplified and unified manufacturing
  - hardware stop

## Controls
### In Progress
- [ ] Toki-san catchup w/ Thomason and Ishida-san
- STM32
- MuJoCo to ROS2 control integration
- URDF corrected
  - reference other humanoids, use standard direction
- Autocalibration
- Document on 5Mbps to 1Mbps (reversible) @1lokeshpatel 
- ROS2 controls for gravity comp: Thomason -> Ishida
- Vanilla Teleop: Thomason -> ?
- ROS2 controls
  - safety filter (prevent large jump of values)

hiroyams, 2025-04-24:
# OpenArm v1.0

hiroyams, 2025-04-24:
[image]

thomasonzhou, 2025-04-24:
In Progress: Add URDF to website